TinyCamML/TinyCamML #1: Calculate power budget

SUPScientist, 2024-01-16:


SUPScientist, 2024-07-17:
Copying screenshot from @efarq's email on 16-Jul-24 titled "TinyCamML Update"
[image]
Image shows battery voltage in far right-hand column, very last value. In 2.5 hours, voltage goes from 4.28 to 4.21. Based on photos in same email, TinyCamML was set up inside, so no solar charging would have taken place. This rate of discharge seems very high and will need to be watched closely.

Options:
1. Reduce sample frequency. Currently 1 sample every 6 minutes (10/hr). 
2. Look into other ways of reducing power draw when not sampling. E.g., cellular standby mode as configured here: https://github.com/TinyCamML/Boron-and-OpenMV/blob/1b7a30e41cd7efd28e730982d60f455be0a2741b/Firmware/Boron/src/TinyCamPublish.ino#L161.

ebgoldstein, 2024-08-12:
(for those reading this comment, please look at #5, which has some detail about optimizing sleep state. )