TinyCamML/TinyCamML #6: Not registering Flood or No Flood

BentleySettin, 2024-03-12:


SUPScientist, 2024-03-12:
Can you be more specific about 

> I plugged both into USB (removing the connecting power wire)

That is, do you mean you used one USB cable per device? 

Did you try the new solar panel + battery to supply 5 V to both (and especially the OpenMV which has higher voltage needs than Boron)?

BentleySettin, 2024-03-13:
Here is what I am seeing where there is no report of flood or no flood: 

[image]

Here is what I am seeing when there is a report: 
[image]

BentleySettin, 2024-03-13:
> Would be helpful to see a copy and paste of the output you're observing in the VS Code serial monitor to have a concrete example of what you mean. Here's the data column in the google sheet:
> 
> 946684803,Flood,4.07,1.33 1710256323,Flood,4.00,2.34 1710256683,,3.98,2.51 1710257044,Flood,3.98,2.68 1710257404,Flood,4.11,2.94 1710257763,,3.98,3.28 1710258123,,4.02,3.87 1710258483,,4.04,3.96 1710258843,,4.02,4.04 1710259203,,4.04,4.13 1710259564,,4.02,4.30 946685174,Flood,0.05,88.50 Format: UNIX Time, flood [or no flood], battery voltage, charge percentage [unreliable]
> 
> Note that some rows show Flood while some rows show nothing where there should be a flood/no flood response.

The last flood you see in the google sheet is from me pressing restart on the Boron. I know we discussed this would not be helpful because it would cause a total reset and not collect real time but I used it to see if it would help report Flood/No Flood and it did. A manual reset that helped both devices communicate on the "same page" makes me wonder if there is a stacking of the communication that disrupts the entirety of UART communications. As said before this could still be rooted in a power supply issue.

SUPScientist, 2024-03-13:
Ah, this might be the clue! Not sure, but this part  

> A manual reset that helped both devices communicate on the "same page" makes me wonder if there is a stacking of the communication that disrupts the entirety of UART communications.

seems especially interesting. I hadn't noticed this pattern because it just looks random to me, but if it often starts with a good UART communication and then falls apart over time, it could be related to the handling of the serial communications more than the power (though could still be the latter).

See this code chunk from a somewhat similar system (Boron communicating with peripheral sensor via UART) where I end and then restart serial communications every time through the loop. It seems like overkill (and might not be optimal coding), but it solved a similar issue. Give it a shot and test again for a similarly long period to see if you can achieve reliable serial comms.

https://github.com/SUPScientist/CellpHOx/blob/77a79682f5c87aff60315a78a794554290044e75/10_CellpHOx_Vipul_PJBImproveTiming/src/10_CellpHOx_Vipul_PJBImproveTiming.ino#L113-L117

BentleySettin, 2024-03-14:
@SUPScientist 

I added this code and updated it [here](https://github.com/TinyCamML/Boron-and-OpenMV/commit/a73a2d89ecc041749dbf780870dfa5da932aa2f8)

I had responses for the first few cycles and then it is has returned to no reporting. 
[image]

BentleySettin, 2024-03-14:
I returned to have a USB powering both devices to continue to refine the problem. I have noticed a new pattern on the openMV. Every 4-5 cycles the openMV flashes green a few times then stops. Following this event the Boron skips a report. I am looking more into this green flashing as a possible issue. 

@ebgoldstein any ideas? Based on the web it seems to be the bootloader.

BentleySettin, 2024-03-22:
The problem has become more and more common of no reporting while connected to the solar panel. When I attach both devices to USB I have no issues at all. When I attach the solar panel and battery to Boron the OpenMV will keep a white light on which seems to be related to power based on openMV forums. 

Based on what I am seeing there seems to still be a power issue.

BentleySettin, 2024-03-22:
Woohoo another issue... 

I left both plugged in to USB running for about 30 minutes and then recieve a white flashing from the openMV. 
When I connect Boron to usb + battery the openMV is unresponsive still. 

I have done some research on this. Multiple people seem to struggle with openMV for more than 30 minutes of running. I am looking into possible updates for the camera, however, am a bit tentative because in one of the forums the creator says       "given how small our operation is some firmware updates brake things. But, generally you want to update the camera". I am still trying to figure out what all these LED flashing or not flashing means based on each scenerio. 

@SUPScientist any ideas?

SUPScientist, 2024-03-22:
If you believe it's a power issue, I recommend checking the charge of the battery attached to the solar panel. 

Recommended steps:
1. Use multimeter to determine battery charge
2. Use multimeter to determine VUSB charge (i.e., voltage going from Boron to OpenMV)
3. If battery is depleted, you may have to manually charge it. Read up on the battery/charger (https://voltaicsystems.com/5-watt-5-volt-35wh-solar-system/) and find a good/safe way to charge it. I imagine you can charge it directly via USB from, e.g., a computer or AC/DC adapter brick (such as standard phone charger).

SUPScientist, 2024-03-22:
> Woohoo another issue...

Please put new issues in a new Issue ;)

BentleySettin, 2024-03-27:
Solar panel charged and now registering flood/no flood with consitency. The lack of reporting from the OpenMV was a battery issue solved by charging the solar panel.

SUPScientist, 2024-04-02:
@BentleySettin, saw you closed this last week and described the resolution. Then saw you reopened it! Is it not resolved? If not resolved, please provide:
- How it should work
- How it is working instead
- Hypothesized root cause of the problem and how you will try to troubleshoot/resolve it.

BentleySettin, 2024-04-03:
I should be consistently have a report of flood or no flood but I am currently seeing sporadic reports. I gave it some time and am realizing it may "just be" a cellular issue. Boron flashes green for a while then will either flash the light blue once or twice or just time out of trying to find cellular. 

I moved the devices to find better connection

BentleySettin, 2024-04-03:
@SUPScientist I am recieving no information on the google sheets even though I am watching the devices turn on and off. 

I am seeing a variety of LED patterns such as; Boron flash yellow, then light blue, then green. 

Could there be something with the antenna?

BentleySettin, 2024-04-03:
Another update: 

I plugged openMV into computer to check out file folder and see if any possible updates because I keep recieving a white flashing light every 30 minutes. On the file folder there was this added page that said " FATAL ERROR:
MemManage ". 

Trying to determine how to deal with this image.

SUPScientist, 2024-04-03:
> I am seeing a variety of LED patterns such as; Boron flash yellow, then light blue, then green. Could there be something with the antenna?

Boron LED status codes are described here: https://docs.particle.io/troubleshooting/led/boron/. Blinking yellow should only happen when you're flashing code or major errors: https://docs.particle.io/troubleshooting/guides/device-troubleshooting/device-blinking-red-yellow-or-frozenno-led/.

ebgoldstein, 2024-06-12:
i think we can close this, as it was likely related to memory growth (#14) when device was run for 30 mins or more. I have a fix for that now, which is committed to github (https://github.com/TinyCamML/Boron-and-OpenMV/commit/730f992ca492b153a643b581057a881996611d9b)... please reopen if needed, but i think we can close...

closed 2024-06-12 by ebgoldstein