TinyCamML/TinyCamML #8: Design enclosure

SUPScientist, 2024-04-02:
### Need
Design housing/enclosure for Boron, OpenMV, internal lipo battery (if needed), and external solar panel/battery.

### Steps
- [ ] Select polycase with dimensions to fit all parts
- [ ] Design placement for penetrators/cable glands on polycase
- [ ] Design placement for camera and window/sun glare shield if needed
- [ ] Design bracket to mount polycase and solar panel together

SUPScientist, 2024-04-02:
Some reference materials exist in related repo, such as a bracket for Voltaic panel/battery and polycase: https://github.com/COAST-Lab/Open-Water-Level/tree/main/Fabrication/Enclosure%20and%20Assembly/Voltaic-Polycase-Bracket. Note that bracket was designed for printing on Form 3 and was made shorter than might be desired given printer limitations. Designed for polycase here https://github.com/COAST-Lab/Open-Water-Level/blob/main/Fabrication/Datasheets/polycase_2d_drawings.pdf and Voltaic bracket here https://github.com/COAST-Lab/Open-Water-Level/blob/main/Fabrication/Datasheets/Voltaic_Bracket_BK102_dwg.pdf.

SUPScientist, 2024-04-02:
@ebgoldstein, I tried to assign this to @jacobstasiewicz but he's not a member of the org and I'm not an owner/don't have permissions to add new members. Could you pull him into the org?

ebgoldstein, 2024-04-02:
@jacobstasiewicz, I'm looping you into this issue re: designng enclosure for the openMV+particle system

To make everyone aware, Jacob and I discussed:
- we need a hole for a cable gland,  but we might not need other cuts.
- 3D printing a mounting system
- I will design a PCB with female headers for the 2 boards to pop into

We discussed options for the camera field of view:
- do we want a clear lid or solid lid?
- do we want a camera dome?
- do we want a hile& visor

ebgoldstein, 2024-04-09:
finding or deisgn/printing a bracket connecting polycase to solar panel (https://github.com/COAST-Lab/Open-Water-Level/blob/main/Fabrication/Enclosure%20and%20Assembly/Voltaic-Polycase-Bracket/Polycase-Voltaic-Bracket%20v12.stl)

SUPScientist, 2024-04-09:
If this design serves as the basis for a new design for this TinyCamML project, please let me know early in the design process. 

Two improvements needed to that design:
1. Slots for mating the solar panel bracket with the custom 3D printed bracket referenced above could be shifted a couple mm for better alignment 
2. It'd be valuable to have material under the top mounting holes on the solar panel bracket to avoid tightening a hose clamp in the top to the extent that it pulls the top of the solar bracket flush with a tree which would result in a non-perpendicular camera angle

ebgoldstein, 2024-04-23:
@jacobstasiewicz:

can you commit the design files and a draft BOM to a new folder in this repository called 'hardware' then close this issue?

I will open a range of issues related to hardware design based on our discussion today, they will all be tagged with 'enclosure' and 'enhancement'

ebgoldstein, 2024-06-12:
this is done... #22 is focused on uploading files

closed 2024-06-12 by ebgoldstein