TinyCamML/TinyCamML #9: Monitor external solar panel/battery voltage

SUPScientist, 2024-04-02:
### Need
Want to be able to monitor voltage of external solar panel/battery which is plugged into Boron's V_USB via the micro-USB socket. 

### Challenge
Particle has methods for monitoring VBATT where a LiPo battery would be connected (via the FuelGauge methods: https://docs.particle.io/reference/device-os/firmware/#fuelgauge) but unclear how to monitor V_USB. Assumption may be that V_USB comes from a stable power supply and does not need to be monitored.

### Steps to Solve
- [ ] Read Particle docs
- [ ] Check Particle forum/post
- [ ] Could feed into a Particle GPIO via voltage divider and use analog input if needed, but this wastes power

SUPScientist, 2024-04-02:
@efarq, can you check this out?

SUPScientist, 2024-04-02:
Here's a reference design for a voltage divider that Particle uses to measure the battery voltage on an Argon: https://docs.particle.io/reference/device-os/api/battery-voltage/battery-voltage-photon-2/#battery-voltage-argon. The voltage on VUSB on the Boron will be higher but the nominal resistances used here are helpful nonetheless. At ≈ 1 MΩ, current wasted will be ≈ 5 µa which isn't too bad.

ebgoldstein, 2024-04-03:
fuel gauge does not seem to work with usb input: https://community.particle.io/t/fuel-gauge-reading-on-the-usb/62568/3

SUPScientist, 2024-04-03:
Yes, I didn't think the fuelgauge would work but it was worth double-checking. See the voltage divider reference above for a proposed solution. I think it will be critical to have solar panel/battery monitoring ability and a voltage divider is very easy to implement and for @ebgoldstein to include in the PCB.

SUPScientist, 2024-04-09:
@efarq to design/test voltage divider to scale 5 V nominal solar panel output down to 3.3 V max acceptable Boron ADC analog input. Using equation Vout = Vin*(R2/(R1+R2)) in standard voltage divider circuit, R2 to be ≈ 2 MΩ and R1 to be rounded up to nearest available resistor that keeps Vout ≤ 3.3 V.

efarq, 2024-04-09:
Just following up - I think I got it worked out! With Phil's help, I found that R1 = 1.125 megaohms, so I rounded it up to 1.3 megaohms, and R2 = 2 megaohms. Then I set up the Boron with some resistors (photo and circuit diagram attached) and ran the code on it. The Boron V(out) was consistently reading about ~2.7V while it was receiving usb power from my laptop. I checked with the multimeter, and that matched what the Boron was saying too. 

I'm uploading the script folder to the Firmware folder, and the Boron+resistors+breadboard are on the counter in Phil's lab. Let me know if anything looks like it needs to be changed. 
[image]
[image]

SUPScientist, 2024-04-09:
Great, thanks, @efarq. Nice work. I see the test script in your commit here https://github.com/TinyCamML/Boron-and-OpenMV/commit/e2b713a82b8a3a851c1bc34a9584dc5fcb082bdf and in the new subdirectory here https://github.com/TinyCamML/Boron-and-OpenMV/tree/main/Firmware/TestingVoltageSplitter. 

To close this issue, I recommend the following: 
1. Create a branch of this repo
2. Integrate solar panel/battery voltage monitor firmware into main Boron firmware in that branch
3. Test new firmware
4. Iterate between 2<—>3 as needed
5. Once firmware works as desired, merge branch back into main.
6. Close issue with link to merge

(@ebgoldstein, workflow seem appropriate to you? As we add teammates here, I'm more than open to thinking through desired github workflow together/following your lead.)

SUPScientist, 2024-04-24:
Great! That looks good at a first glance. Confirming that you tested this on the hardware? If so, please say so and click "close with comment." 

I recommend removing the percent charge variable and report. Battery charge is non-linearly related to voltage (google "battery discharge curve" for visuals) and can easily be inaccurate. Voltage will suffice. No branch/pull request needed for that change; just remove relevant parts of .ino file and recommit/push, please.

efarq, 2024-04-26:
Yes, I tested it on the Boron and OpenMV that bentley was using, and it seems to work.

efarq, 2024-06-25:
I'm reopening this so I remember to look at it this week. I just need to double check the code/voltage with a multimeter for this because it's still reporting about 2.8-2.9 V when plugged into usb power. I thought I scaled it up properly by multiplying by an extra ((R1 + R2) / R1) 

https://github.com/TinyCamML/Boron-and-OpenMV/blob/3a03ea4b75fd1438a5236e86fadcb9e209097110/src/boron/src/TinyCamPublish.ino#L72

SUPScientist, 2024-07-17:
Solved with https://github.com/TinyCamML/Boron-and-OpenMV/commit/8a047e3b0cbdd27e7818fff782918f5fbd4ddb85

efarq, 2024-07-22:
I'm actually not 100% convinced this is truly resolved. So currently, I have the TinyCamML running off of my computer usb, so it should be reading pretty close to 5V. However, it is reporting values around 4.36V, suggesting maybe my scaling is off? Any ideas of things for me to review? 
[image]
[image]

SUPScientist, 2024-08-12:
Reopening based on @efarq's comment above and preliminary data showing no apparent trend in voltage (screenshot below from analysis in https://github.com/TinyCamML/Boron-and-OpenMV/blob/main/analysis/tinyCamML_analysis_20240812.ipynb).

[image]

Notes:
- Unclear if data show anything unexpected after relatively short deployment
- Hypothesis: a voltage regulator on Voltaic panel/power supply may "try" to supply 5 V no matter what the battery's actual charge is and we may not be able to monitor this after all. Unsure, but more investigation needed.

efarq, 2024-09-27:
Closing this now after talking to Voltaic Systems and having them tell us there is no way to monitor battery voltage from the solar panel, likely due to a voltage regulator

SUPScientist, 2024-09-27:
Seems appropriate to close; I just referenced this in a new issue at https://github.com/TinyCamML/Boron-and-OpenMV/issues/43 so that this idea, even if not the solution pursued in this issue, is retained.