TinyCamML/TinyCamML #11: Streamline repositories

ebgoldstein, 2024-04-03:
There are 8 repositories in this organization, but i am not sure if we need them all. Can we remove/reduce/combine repositories if they are not critical or if they are redundant?

- `.github`- keep, to explain the Github org
- `OMVmodel` - keep, this is the code to make the openMV model (but we can remove the openMV code, since it might be duplicate with the OpenMV code in `Boron-and-OpenMV`?
- `Boron-and-OpenMV` - the one we all seem to be working in, which seems to be critical
- `TinyCamPublish1` - needed?
- `Boron` - is this needed? or does `Boron-and-OpenMV` supersede it?
- `Example-Repo` - needed still?
- `Boron-and-OpenMV-Basics ` - is this needed? or does `Boron-and-OpenMV` supersede it?
- `Boron` - is this needed? or does `Boron-and-OpenMV` supersede it?
- `Honors-Thesis-Requirements ` - are you using this? if not, lets delete... 

(cc: @SUPScientist , i want your opinion/perspective in case i am missing something)

BentleySettin, 2024-04-04:
Sounds good. I can delete the other ones.

ebgoldstein, 2024-06-12:
archived repositories, and then adjusted the org readme

closed 2024-06-12 by ebgoldstein