TinyCamML/TinyCamML #23: send datetime from Boron to OpenMV

ebgoldstein, 2024-06-13:
this is likely a uartread w/ OpenMV

ebgoldstein, 2024-06-13:
i assigned this to @efarq , but I am also helping with this

efarq, 2024-06-21:
For the particle boron code, I added a Serial1.println(real_time) after the Serial1.readstring() so that it would print its datetime to the OpenMV. 

For the openMV code, I added a curr_time = str(uart.read()) to after the uart.write() function, so it would read the datetime for the boron. 

Currently, this is not working. I am wondering if there needs to be something like a serial1.end and begin to reset the channel before trying to exchange the datetime. Or perhaps an "if serial1.available()" statement before it tries to send the datetime. Any other ideas @ebgoldstein @SUPScientist ?

SUPScientist, 2024-06-21:
First, re: process: can you create a branch (maybe you already have), commit code, and add a direct permalink to the relevant chunk(s) in this issue here? 

Also, the more description about what is expected behavior vs. what you're seeing (not working), the better. 

Second, re: more immediately relevant ideas, you could be right about a well-placed `Serial1.end()` and/or `Serial1.flush()` idea on the Boron side. On the OpenMV side, given that you're using `println()`, you might want `uart.readline()     # read a line` https://docs.openmv.io/library/pyb.UART.html

SUPScientist, 2024-06-21:
I bashed my head against the wall with a similar goal a few years ago. Thankfully I made a repo at that time and posted the code here: https://github.com/SUPScientist/CellpHOx/blob/master/10_CellpHOx_Vipul_PJBImproveTiming/src/10_CellpHOx_Vipul_PJBImproveTiming.ino. 

In that case, I did a write and then a read on the Boron side (https://github.com/SUPScientist/CellpHOx/blob/77a79682f5c87aff60315a78a794554290044e75/10_CellpHOx_Vipul_PJBImproveTiming/src/10_CellpHOx_Vipul_PJBImproveTiming.ino#L120-L123) whereas you're doing the reverse but it shouldn't matter much. There's nothing fancy in those couple relevant lines of code, but note the use of timeouts on the Boron side, which is buried in the way it listens to the peripheral sensor and waits a set amount of time for the response. Maybe the OpenMV has something similar.

efarq, 2024-06-25:
#commit 3a03ea4 . I believe I got this working. It was Serial1.print() for the boron, refreshing the serial1 in between, and uart.read() for the openmv

closed 2024-06-25 by efarq

ebgoldstein, 2024-06-25:
Great work @efarq !!