TinyCamML/TinyCamML #34: Carrier Board / PCB for Boron and OpenMV

ebgoldstein, 2024-08-07:
Current status:
- Thanks to hard work from @efarq , we now have a stable prototype some new bits of hardware (i.e., a MOSFET - thanks for the suggestion @SUPScientist !). The wiring, microcontrollers, and cell antenna fit in the polycase, and are mounted using 3D printed parts (thanks @jacobstasiewicz ). 

In the future, it would simplify future builds if there was a manufactured PCB where the electronic parts were pre-populated on the board, and the microcontrollers could just plug into headers. to do this we would likely need:

- A PCB design that accounts for the microcontroillers and the electronic components (see the circuit diagram)
- new 3d printed parts that attach the PCB to the polycase

mjsaenz, 2024-12-17:
Hi everyone! I am Matthew Saenz. I am a master's student in Electrical and Computer Engineering at Carnegie Mellon University and an intern at the Field Research Facility in Duck, NC. Evan has brought me on to work on this task, as well as investigating the solar power supply.
I'd like to begin with listing high level system requirements I derived from my discussion of the project with @ebgoldstein this morning:

1. The PCB shall have mounting locations for all devices in the circuit diagram, i.e. the Boron board, the OpenMV board, the MOSFET, and the voltage divider resistor.
2. The PCB shall connect the devices as specified in the circuit diagram via circuit traces.
3. The PCB shall not displace the antenna mount location, but may modify the other mounting hardware as necessary.
4. The PCB shall include a mounting location for a screwhead terminal or similar to enable testing alternative power supplies.
5. The USB and SIM card ports of the Boron board shall be easily accessible in the fully assembled configuration.
6. The SD card port of the OpenMV board shall be easily accessible in the full assembled configuration.

Regarding the placement of the PCB, would it be better to mount it on top of the Boron and OpenMV, or below them? I am currently leaning towards below the boards, as it would enable a simpler form factor for the PCB reducing manufacturing costs, but in the baseplate assembly drawing, the header pins appear to be oriented upward from the boards.

Please let me know if you have any feedback on these design requirements, as well as if there are any other desired features of the PCB.

SUPScientist, 2024-12-17:
Great to meet you, Matthew, and welcome to the team :) In case @efarq doesn't have automatic notifications, I'm tagging her here explicitly as she has spent the most time of any of us with hands on the device. @efarq, maybe you could share your google slide deck with @mjsaenz as well as it has some helpful photos (while you're at it, could you add a couple of those to the assets folder here? 

Responding to your system requirements below where there's anything to add. If I haven't added anything, then I agree!

1. The PCB shall have mounting locations for all devices in the circuit diagram, i.e. the Boron board, the OpenMV board, the MOSFET, and the voltage divider resistor. *PJB response: we haven't been able to use the voltage divider for meaningful voltage monitoring. @efarq please confirm but I recommend removing unless we anticipate future usability. FET switch (with its resistor definitely stay.*
2. The PCB shall connect the devices as specified in the circuit diagram via circuit traces.
3. The PCB shall not displace the antenna mount location, but may modify the other mounting hardware as necessary. *PJB response: please see https://docs.particle.io/hardware/certification/antenna/#cellular-antenna-dimensions for antenna spec. @jacobstasiewicz please confirm that current antenna mount meets spec.*
4. The PCB shall include a mounting location for a screwhead terminal or similar to enable testing alternative power supplies.
5. The USB and SIM card ports of the Boron board shall be easily accessible in the fully assembled configuration. *PJB response: Boron uses a soldered on eSIM so access is not required.* 
6. The SD card port of the OpenMV board shall be easily accessible in the full assembled configuration.

Regarding the placement of the PCB, would it be better to mount it on top of the Boron and OpenMV, or below them? I am currently leaning towards below the boards, as it would enable a simpler form factor for the PCB reducing manufacturing costs, but in the baseplate assembly drawing, the header pins appear to be oriented upward from the boards.
*PJB response: Boron has male header pins pointing downward (such that your carrier PCB should be underneath Boron with female headers/sockets). I thought this was also the case for OpenMV but a recent photo from @efarq shows the opposite, so we need her confirmation.*

Other desired features. *PJB response: the current connector that accepts the solar panel cable's plug and converts it to micro-USB (this: https://voltaicsystems.com/A101/) is a little clunky in my opinion, but it works. Would like @efarq's opinion here also (if it ain't broke, don't fix it?) but the power jack socket on this board (https://github.com/iorodeo/iorodeo_feather_tripler_enc1) that then runs the 5V trace to the Boron's VUSB input could also work. Researcher needs to remember to unplug solar before using USB cable from computer or vice versa.*

ebgoldstein, 2024-12-18:
Hi @mjsaenz , thx for the list. 

@SUPScientist - this morning @mjsaenz and I had a call to discuss these issues, and he is connected to @efarq via email to get an actual device. Jacob is away for the week, but the antenna works so my opinion is that we not mess with the configuration. 

the carrier board should attach from below. I can confirm via a phone call w/ @efarq that the OpenMV @mjsaenz will get has downward facing male headers.

Power:
- I spoke to @mjsaenz about power, and trying to estimate power budget, which we have discussed in variou issues: #1, #9, #10, #43, and #45 .
- I mentioned that the voltaic panel and battery we are sending over is regulated, and will not output meaningful numbers for power budget estimation.
- I mentioned that Particle boards expose a way to monitor battery voltage from a LiPo via the C++ API (https://docs.particle.io/reference/device-os/api/fuelgauge/getvcell/), so we are interested in working with a new battery that could integrate with this ability.
- I mentioned that we want to future proof this PCB by allowing for other battery and solar panel solutions to be connected, and monitored (so using ones that are not regulated). This might necessitate adding or retaining things on the PCB (voltage divider, screw terminals, etc).
- the power management stuff is definitely a 'stretch goal', so my opinion is that we ask @mjsaenz to focus first on just building the basic PCB, and then if he has time tackle some of the power stuff.

how does this sound?

SUPScientist, 2024-12-18:
Nope, shouldn't mess with the antenna holder itself, agreed. The important spec to bear in mind for PCB design is this: 
> All portions of the antenna should maintain proper spacing from electronics, grounded metal, or active metal.
Recommended: 12mm
Minimum: 8mm

I didn't mean we should mess with antenna holder, just that we should maintain minimum distance between antenna and other electronics.

As for monitoring battery voltage, what we do on the water level sensors is (1) use the same Voltaic panel/battery as TinyCamML and (2) also use a small LiPo inside the polycase and the code: 
`// Get battery charge 
    float cellVoltage = batteryMonitor.getVCell();
    float stateOfCharge = batteryMonitor.getSoC();`

ebgoldstein, 2024-12-18:
so there are two batteries - the voltaic one and then a smaller LiPo? is the idea that the small battery is just extra, and will better reflect the state of the charge for the both batteries?

efarq, 2024-12-18:
Hi all,

I have indeed shared the TinyCamML google drive with Matthew, so let me
know if you don't receive it. Also, I realized the current configuration of
the OpenMV has the female headers facing down (opposite of what I told you,
Evan). However, this could be easily changed for all future TinyCamMLs, as
each OpenMV comes with male headers that can be soldered on instead of the
female ones.

I am going to mail you the TinyCamML tomorrow, Matthew, so I can get
together a few extra items to include. I just bought a fresh openMV as well
to ship to you, so that you can have the correct male header configuration
so you can move forward with a pcb design where the Boron and OpenMV just
"plug in". Do you have access to a soldering iron to solder on the male
headers when it arrives?

In response to some of the other inquiries above:
1. We haven't been using the voltage divider, so that can be excluded in
the initial PCB design, unless you find a way to incorporate it for voltage
monitoring later on. Sounds like there are other options though for voltage
monitoring, such as including a mini lipo battery inside the casing.
However my question about this is, would we still be able to use the VUSB
pin on the Boron if a battery is plugged in?
6. The microusb needs to be easily accessible on the OpenMV as well.

I think I like the idea of sticking with the
https://voltaicsystems.com/A101/ part for now until we figure out a
solution for decreasing solar panel size and/or maybe getting rid of the
external battery?

Best wishes,

Liz Farquhar

SUPScientist, 2024-12-18:
> so there are two batteries - the voltaic one and then a smaller LiPo? is the idea that the small battery is just extra, and will better reflect the state of the charge for the both batteries?

Correct, @ebgoldstein. Voltaic panel/battery pack @ 5V is hooked up to VUSB on Boron and VIN on OpenMV, both of which can handle 5V. A smaller lipo @ 3.7V nominal is hooked up to Li+ on Boron. The Boron's PMIC draws from VUSB and charges the lipo if there's extra juice. See fig below; lipo stays fairly close to full charge for months.

> However my question about this is, would we still be able to use the VUSB pin on the Boron if a battery is plugged in?

Yes, @efarq, also described above. The Boron's PMIC handles current flow. Something would likely break if we had two things plugged into Boron's VUSB, such as a computer connected via micro-USB cable + voltaic battery wired to VUSB via a trace on a new PCB.

This figure shows an almost yearlong time-series of two water level sensors that measure/report hourly. The first few weeks used an internal lipo as described above with a solar panel **without** an extra battery pack plugged into VUSB. Swapped out the battery packless solar panel in April for the identical Voltaic panel/pack we're using here. Battery V shown here is that of the internal small lipo.
[image]

mjsaenz, 2024-12-20:
Thank you all for the input! I apologize for the delay in replying here, I am getting caught up after driving home all day Wednesday. I have updated my design requirements for the PCB as follows (changes **bolded**):

1. The PCB shall have mounting locations for all devices in the circuit diagram, i.e. the Boron board, the OpenMV board, the MOSFET, **~~and the voltage divider resistor~~**.
2. The PCB shall connect the devices as specified in the circuit diagram via circuit traces.
3. The PCB shall not displace the antenna mount location **in order to maintain the recommended distance from electronics of 12mm**, but may modify the other mounting hardware as necessary.
4. The PCB shall include a mounting location for a screwhead terminal or similar to enable testing alternative power supplies.
5. The USB **~~and SIM card~~** port**~~s~~** of the Boron board shall be easily accessible in the fully assembled configuration.
6. The **microUSB and** SD card port**s** of the OpenMV board shall be easily accessible in the full assembled configuration.
**7. The OpenMV board and Boron board shall be mounted above the PCB, i.e. "slot-in."**

As for your question @efarq, I unfortunately will not have easy access to a soldering iron while at home,, but i will reach out to some friends in the area who may have their own.

SUPScientist, 2024-12-20:
Nicely done distilling this thread. Looks good to me!

mjsaenz, 2024-12-21:
Question for the MOSFET and its resistor, do we want to continue to use the through-hole components listed in the BoM, or would we want to change to surface mount components?

mjsaenz, 2024-12-21:
Additionally, would it be desired for the Boron and OpenMV boards to be directly soldered into the PCB, or mounted into female header pins on the board. Directly soldering would allow for the overall boardstack to be shorter by trimming the extra pin length, but would make substituting/replacing the boards more difficult.

ebgoldstein, 2024-12-22:
My preferences are: 
- through-hole components for ease of soldering. 
- boards mounted via female headers for easy replacement. (As long as it fits in the enclosure)

mjsaenz, 2024-12-29:
Hi all,
I have begun digging into the schematic layout for the PCB and wanted to clarify the role of the mosfet in the circuit schematic, as its source is connected to the GND pin of the boron board and its drain the GND pin of the OpenMV. Does the switch provide a form of voltage regulation, and if so, should the GNDs of the two boards not be tied together?
I also wanted to note that for the alternative power supply (requirement 4), this will have to be a non-standard voltage, as the boron board only accepts an input voltage in the range of 3.6-4.2 VDC. This is within the input range of 3.6-5V of the OpenMV's Vin pin.

efarq, 2024-12-29:
Hi, The mosFET switch is only used to turn on and off the power to the OpenMV, which was necessary because otherwise the OpenMV gets “out of sync” with the Boron. I’ll try to find the documentation that informed why I set up the mosfet that way. I believe the Boron also accepts ~5V voltage in its microusb port. Best wishes, Liz Farquhar

mjsaenz, 2024-12-30:
Thanks for the clarification on the mosFET @efarq ! Any documentation you have on it would be appreciated.
As for the second point, while the Boron does take in 5V via the microUSB port and outputs it at 5V on the VUSB pin, for its use, it is regulated down to 3.3V by a step-down regulator. The Li+ pin is the only pin which functions as a voltage in pin on the boron board, which accepts an input voltage in the aforementioned range of 3.6-4.2 VDC. I realized almost immediately after stepping away from the computer that we could also achieve this via a voltage divider, allowing the OpenMV board to receive its full 5V input, and the Boron board to receive an input voltage within its accepted range.
Here is the power documentation for the boron board: [https://docs.particle.io/reference/datasheets/b-series/brn404x-datasheet/#power](url)

efarq, 2025-01-02:
Hi - so I had originally used this mosFET switch carrier board from Sparkfun: https://www.sparkfun.com/products/retired/12959 , which shows you exactly what to connect to the "device" (OpenMV in this case) and the "system" (the Boron). Using the schematics listed in their diagrams and this conversation (https://github.com/TinyCamML/TinyCamML/issues/27#issuecomment-2246320316), we were able to recreate it just by putting it on our own tiny board.

mjsaenz, 2025-01-02:
Following the discussion on that thread, it is my understanding that the FET is necessary to restart the OpenMV every publish cycle of the Boron board in order to prevent the image not saving error which occurs if the OpenMV runs for too long (> 40 minutes). The A0 pin essentially acts as an enable pin for the OpenMV, as when it is HIGH, it pulls the GND pin of the OpenMV board low by connecting it to the GND pin of the Boron board, and therefore the ground of the power supply.

I have made a quick mock-up of the updated circuit diagram to accommodate an alternative power supply. It uses a 3 pin header with 2-pin jumper to switch the input voltage to the OpenMV board between the VUSB pin of the Boron board and the 5V input pin of the screw terminal. I have left the resistors of the voltage divider to step down the screw terminal voltage from 5V to the accepted range of the Li+ as a black box for now, as this depends on the input current of the power supply and desired current at the Li+ pin. The Boron's voltage regulator has a 1000mA maximum, but no current limit is listed for the Li+ pin.

[image]

ebgoldstein, 2025-01-06:
this looks appropriate and good to me, but I'm interested to hear from @efarq and @SUPScientist ..

SUPScientist, 2025-01-07:
Thanks, @mjsaenz. We have a few modification requests, sketched up and described below.

[image]

Please add to the PCB the following: 
1. J1: Female socket/barrel plug connector to accept the plug from the voltaic battery pack (voltaic's output is Male 3.5 x 1.1mm output cable: https://voltaicsystems.com/5-watt-5-volt-35wh-solar-system/).
2. SW1: On/off switch immediately after J1.
3. Short VUSB on Boron, VIN on OpenMV, and output of switch SW1 (all 5V).
4. Female header pins to accept Boron and OpenMV and necessary traces as indicated (e.g., Rx/Tx to P1/P0)
5. FET as shown with appropriate traces.

Note I added the GND nets to indicate shared ground between output of the voltaic battery pack and Boron. OpenMV is only grounded when FET switch utilized thus controlling its power. 

This sound right, @ebgoldstein @efarq? Thanks!

mjsaenz, 2025-01-07:
@SUPScientist Understood, I will send an updated board schematic soon. To confirm. we are no longer interested in pursuing alternative power supplies via a screw terminal?

SUPScientist, 2025-01-07:
Thank you! Confirmed (assuming you can readily find the proper mating female barrel plug connector which will serve that same purpose).

mjsaenz, 2025-01-07:
My branch can be found here: https://github.com/TinyCamML/TinyCamML/tree/PCB

SUPScientist, 2025-01-07:
Wow, that was fast, thanks! Looks great! And as @ebgoldstein nicely stated earlier, thanks for putting up with our evolving whims.

I think it's good to go and I have no requested changes, just gonna jot a couple notes to make sure we're all on the same page.

@efarq, are you good with orientation of Boron & OpenMV? This shows micro-USB connector facing bottom of board on Boron, camera facing top of board on OpenMV with its micro-USB connector on bottom of board. Seems good to me and mirrors recent photos of your setup but your confirmation would help.

The male barrel plug comes into the connector from the right edge, correct? @efarq, please confirm you're good with that rotation. That might make for a tight fit against the right edge of the PolyCase. Would the plug entering from the bottom edge upwards (i.e., female barrel connector facing down) make for an easier connection?

Finally, to be explicit, I want to write out the function of S1 and confirm I'm understanding. Is this correct?
VUSB (Boron) and VIN (OpenMV) are shorted regardless of switch position.
- With switch in **ON** position, barrel plug's voltage is also connected to VUSB and VIN (i.e., powering both boards). 
- With switch in **OFF** position, barrel plug's voltage is disconnected from rest of circuit.

mjsaenz, 2025-01-07:
> Finally, to be explicit, I want to write out the function of S1 and confirm I'm understanding. Is this correct?
> VUSB (Boron) and VIN (OpenMV) are shorted regardless of switch position.
>
>     With [...]

That is correct.

I also realized that I forgot to specify the units in the board diagram. below is an updated diagram with the units:
[image]

The through-holes on the board are designed to fit M3 screws for mounting within the PolyCase.

efarq, 2025-01-10:
Looks great! Sorry for delayed response (fieldwork), but I think the orientation of the openmv and boron looks great! However, I think the barrel plug and the switch should swap places because we need to be able to access the microsd card on that side of the openmv easily, and its current location is blocking it. I think the barrel plug is fine to continue facing towards the right, since the cable gland is on that side and the cable only needs to go straight from the cable gland to plug in.

mjsaenz, 2025-01-10:
@efarq Understood! I have swapped the locations of the switch and barrel jack to allow easier access to the SD card slot.
[image]