TinyCamML/TinyCamML #35: new way to deal with TF in OpenMV

ebgoldstein, 2024-08-07:
the new openMV firmware has some breaking changes for how it deals with ML models:

https://openmv.io/blogs/news/giant-tensorflow-update

efarq, 2025-07-10:
I've got a pretty good start on this as seen in commit cffa386 . However, I think @ebgoldstein should check to make sure I have correctly designated the first and second elements of the TF_objs list to Flood and No Flood, respectively, and everything else makes sense. Additionally, I have noticed that the memory allocation (which previously ran around 9300) is now running at 2594064 with this new version, so any tips you may know of for decreasing that would be helpful.

ebgoldstein, 2025-07-14:
I think that is correct re: the TF_objs.
I think we need to run the code on teh device as a test re: the memory, to see if it fails at some point when being deployed..

efarq, 2025-07-29:
Just wanted to follow up that I ran the new ML code for 12 days on a TinyCamML and it did indeed capture the expected ~240 images and outputs for each day. It looks like it might have missed publishing a couple times, which is probably more of a Boron/service issue than an OpenMV issue, since everything was logged. Continuing to look into it.