TinyCamML/TinyCamML #36: Assess how physical housing stands up to elements

ebgoldstein, 2024-08-07:
after deployment, we should evaluate this.

ebgoldstein, 2024-09-24:
@efarq - any insight you want to add here? or should this issue be closed because things are working fine so far

efarq, 2024-09-27:
Hi, 
I think we can close this until something specific comes up! It seems like everything looked pretty good to me from the last deployment, although I have the follow notes:

- salt was crusted along the battery, so we should make sure we always put o-ring grease plentifully on the o-ring of the battery (and the polycases')
- the 3D printed parts seemed totally fine from this short deployment, but I'll be interested to see how they do with long-term UV exposure
- the polycase did tilt down overtime (causing the images to get more and more downward facing), suggesting that either I need to tighten the tilt screw harder during deployments, or if it's a continued issue, change the design a bit
- any hardware that wasn't stainless steel rusted 
- we'll see how trusty these plastic cable glands are long-term - so far, so good

closed 2024-09-27 by efarq