TinyCamML/TinyCamML #44: openmv won't save > 1024 images with certain micro-SD cards

SUPScientist, 2024-10-25:
### Observed behavior
@efarq writes in email "TinyCamMLs Maiden voyage" on 23-Oct-24: "So TinyCamML2,3, and 5 stopped saving images once they hit about 1000 images saved (which was like yesterday) but did keep recording in their datalog. I reset the tinycammls so they are still out there running currently and I’ll probably check on them Friday. TinyCamML1 seems to be totally fine, which is interesting because it is the only one whose sd card is different (picture below) and it has the voltage divider. All of the sd card formats are the same, so I’m thinking this sounds like some sort of specific sd card bug? I’m going to mess around with the extra TinyCamML and see if I can recreate it but let me know if y’all have any ideas."

### Expected behavior
TinyCamMLs keep saving images as long as there is available space on micro-SD card

SUPScientist, 2024-10-25:
This thread and in particular [this comment](https://forums.openmv.io/t/consuming-time-issue-of-saving-image-to-sd-card/5815/2) suggests using `ImageWriter` rather than `save`. `save` evidently spends time finding open space on the SD card that increases as more space is filled up, maybe getting to the point that the save operation isn't completing when the device goes to sleep.

Another quick band-aid solution could be to add a delay or query the result of 
https://github.com/TinyCamML/Boron-and-OpenMV/blob/bcfeacb64a2c065acb4354f63079edc074f35b6f/Firmware/OpenMV/PublishOpenMV.py#L74
and have a blocking loop below it that doesn't let you proceed (and go to sleep) until it responds. I don't know if the `save` method returns anything, but that's worth looking into also.

Lots of forum posts about seemingly related issues! https://forums.openmv.io/t/saving-stream-of-jpeg-images-to-sd-card/1648

ebgoldstein, 2024-10-25:
imagewriter is deprecated, so it would be image.ImageIO . this would essentially save a video file, that we would decode later.. this is not that fun, so i suggest a possible hack: new directories every day

so before the while loop:

> def get_today_folder():
>     """Create and return the folder path for today's date."""
>     date_str = utime.strftime("%Y-%m-%d")  # Format: YYYY-MM-DD
>     folder_path = f"./images/{date_str}"
>     if not os.path.exists(folder_path):
>         os.mkdir(folder_path)
>     return folder_path

and in teh loop:

> 
>    # Read current time from UART
>     poll = uselect.poll()
>     poll.register(uart, uselect.POLLIN)
>     poll.poll()
>     curr_time = uart.read().decode('utf-8')
> 
>     # Create a file name using the current time and flood state
>     file_name = f"{curr_time}_{floodstate}.jpg"
> 
>     # Get the path for today's folder and save the image
>     today_folder = get_today_folder()
>     img.save(f"{today_folder}/{file_name}")
> 
>     # Log the data in DataLog.txt
>     with open("DataLog.txt", 'a') as file:
>         log_entry = f"{curr_time},{floodstate},{gc.mem_alloc()}\n"
>         file.write(log_entry)

efarq, 2024-10-30:
Just wanted to let y'all know I am currently testing this out right now and I think its working. I'm going to let it run for a few days at home before I commit the code, but basically I added a one second delay before turning off the OpenMV in the Boron's code, and then in the OpenMV code, I added this function definition, using the Boron's timestamp as the basis: 

`def get_today_folder():
    ts = int(curr_time) #time from boron
    tsdt = list(time.localtime(ts)) #convert from tuple to list
    tsdt[0] = (int(tsdt[0])-30) #subtract 30 years bc micropython starts timestamps at 2000
    date_str = f"{tsdt[0]}-{tsdt[1]}-{tsdt[2]}" #extract date
    folder_path = f"{date_str}"
    if not folder_path in os.listdir():
        os.mkdir(folder_path)
    return folder_path
`
and at the bottom: 
`today_folder = get_today_folder()

    with open("./DataLog.txt", 'a') as file:
        file.write(curr_time + "," + floodstate + "," + str(gc.mem_alloc()) + "\n")

 img.save("./" + today_folder + "/" + file_name + ".jpg")`

closed 2025-01-13 by efarq