TinyCamML/TinyCamML #47: make PCBs

ebgoldstein, 2025-01-10:
- [ ] get PCBs
- [ ] get componenets
- [ ] manufacture
- [ ] test

efarq, 2025-02-02:
So I was just about to test the PCBs, but I ran into one small issue. The measurement for the OpenMV pins is off by a maybe 1 mm or so, so it is not possible to simply plug in the OpenMV. Photos attached. 

[image]
[image]
[image]

ebgoldstein, 2025-02-03:
@mjsaenz - is this is a quick fix that you do you think?

mjsaenz, 2025-02-03:
> [@mjsaenz](https://github.com/mjsaenz) - is this is a quick fix that you do you think?

Hey Evan,
It should be a quick fix. My apologies for this mistake. The schematic on OpenMV website did not have the distance denoted on it, so I had to use the ruler on the CAD model viewer. It looks like the "object to object" option I used measured closest point to closest point, giving me the original 30.78mm distance. Remeasuring with "point to point" shows a distance of 32.26mm center to center. I will update the model and upload it to the repo tonight.

ebgoldstein, 2025-02-03:
No worries at all @mjsaenz !!

ebgoldstein, 2025-02-03:
I just verified that distance with my calipers on an OpenMV. @efarq - do you have calipers handy to verify that measure on your end?

mjsaenz, 2025-02-04:
Hi all,
I have updated the board layout to the proper distance between the OpenMV pins here: 3b9fc03c831bb1989981e911a0714abf54d13449

ebgoldstein, 2025-02-04:
awesome- thx @mjsaenz - ordering next round of boards RN...

ebgoldstein, 2025-02-04:
@mjsaenz - Q for you: are the drill holes and board shape/size the same? 

(I ask because @jacobstasiewicz could still design the 3D printed mount - (he has a v1 boards. )

mjsaenz, 2025-02-04:
> [@mjsaenz](https://github.com/mjsaenz) - Q for you: are the drill holes and board shape/size the same?
> 
> (I ask because [@jacobstasiewicz](https://github.com/jacobstasiewicz) could still design the 3D printed mount - (he has a v1 boards. )

Yes. The only change is the position of the OpenMV pins.

ebgoldstein, 2025-03-24:
@efarq - have you deployed a TinyCamML with the new PCB? Are we ready to close this and order a few more boards?

efarq, 2025-03-27:
The TinyCamML with the new PCB ran successfully from 3/13/2025-3/27/2025 with no issues. Therefore, I think we are ready to close this and order some more boards